pulp-platform/riscv-dbg #3: Executing an illegal instruction as abstract command doesn't get reported back through abstractcs

imphil, 2019-02-19:
We are currently porting our zero-riscy fork to use this debug module. I noticed an issue with OpenOCD not being able to connect and I wanted to double-check with you if you've seen the same thing on your side, if this has never been tested, or if my integration is somehow buggy (which is entirely possible).

Here's the story.

1. During the target examination (i.e. startup), OpenOCD always issues a 64 bit register read through an abstract command to guess XLEN. 
   (Underlying idea: if 64 bit register read succeeds XLEN = 64, otherwise XLEN = 32)
   That abstract command gets translated into an 64 bit LD instruction (0x38853023 => LOAD width = 3, dest = 8, base = 10, offset = 0x380)

    - View from OpenOCD ([riscv-013.c:1507](https://github.com/riscv/riscv-openocd/blob/8dd5d2a71064e8cab577b6681d08314e919af753/src/target/riscv/riscv-013.c#L1507))

        ```c
       int result = register_read_abstract(target, NULL, GDB_REGNO_S0, 64);
       ```

    - View from hardware (custom debug prints)
        ```
        DMI request: addr=0x17 (command), op=0x2 (write), data=0x00321008
        DMI response: data=0x00000000, resp=0x0
        ```

2. The CPU issues an illegal instruction exception and jumps back into non-debug mode
     ```
     Illegal instruction (core 0) at PC 0x1a110348
     ```

3. OpenOCD gets loops while `abstractcs.busy` is set

    - View from hardware:
      ```
      DMI request: addr=0x16 (abstractcs), op=0x1 (read), data=0x00000000
      DMI response: data=0x08001002, resp=0x0
      ```
      response.data means progbufsize = 8, busy = 1, datacount = 2

    - View from software:
      `while(1)` loop in wait_for_idle() @ [riscv-013.c:679](https://github.com/riscv/riscv-openocd/blob/8dd5d2a71064e8cab577b6681d08314e919af753/src/target/riscv/riscv-013.c#L679)

4. OpenOCD reports a timeout and continues when the duration set with `riscv set_command_timeout_sec` is reached (2s by default)

    ```
    Error: Timed out after 2s waiting for busy to go low (abstractcs=0x8001002). Increase the timeout with riscv set_command_timeout_sec.
    Error: Abstract command ended in error 'busy' (abstractcs=0x8001102)
    ```
5. From now on things go downhill: OpenOCD assumed XLEN = 64 and other things, which don't result in working interactions.

Setup:
- OpenOCD from https://github.com/riscv/riscv-openocd (master)
- riscv-dbg from the pulpissimo_integration branch (with custom patches on top)

zarubaf, 2019-02-20:
I can only comment on how it is supposed to work (the same happens when probing a 64 bit core with 128 bit access, e.g.: `lq/sq`):

The core will encounter an illegal instruction and the exception handler should jump to https://github.com/pulp-platform/riscv-dbg/blob/f256f906db8241f7a9e50dab4aae0d498cf42892/debug_rom/debug_rom.S#L29

The corresponding code will let the debugger know that the core encountered an exception and flag the access as an error. As per debug spec p.41:

>Exceptions don’t update any registers. That includes cause, epc, tval, dpc, and mstatus.
They do end execution of the Program Buffer.

So the exception does not alter the architectural state. I am not sure whether this answers your question. Corresponding exception code in Ariane is here:
https://github.com/pulp-platform/ariane/blob/90cebe6e7091a67f2926ed8d4dfdf9f122092f36/src/csr_regfile.sv#L910

imphil, 2019-02-20:
Thanks for looking at that @zarubaf. I've come to a similar conclusion after a night's sleep, it's good to have that confirmed. Since I didn't find code to alter the exception jump target for debug mode in the RI5CY core (where I ported the zero-riscy stuff from) it might just been missing there as well, or I am missing it. Either way, I'm giving it a go and will let you know how it goes.