pulp-platform/riscv-dbg #16: Change to resumereq and resumeack

jm4rtin, 2019-05-14:
As promised when we talked before, the following is the change I made to get the RI5CY working with _SEGGER Embedded Studio for RISC-V_.

When looking at the _RISC-V External Debug Support_ specification (I refer to version 0.13.2 which may have added some clarification on this behavior) in section _3.5 Run Control_ this is a paragraph describing resume. In here it states a write of 1 to _resumereq_ will resume the core. It does not say you have to clear _resumreq_ first. If you look at section _3.12.2 Debug Module Control_ the _resumereq_ field has W1 access (meaning only a write of 1 is valid). This is intended as a one-shot signal and the J-link will not write the field to 0 before issuing another resume request.

The second issue is that when this is fixed the _resumeack_ will go low too soon. This was changed to go low only when a 1 is written into _resumereq_ (as stated in section _3.5 Run Control_).

You can see the changes I made in this branch: [emmicro-us/fix-resumeack](https://github.com/pulp-platform/riscv-dbg/compare/pulpissimo_integration...emmicro-us:fix-resumeack?expand=1)

The branch also includes some sane parameter defaults to play nice with ncsim. Please let me know if you want me to open a pull request.

bluewww, 2019-05-15:
Thanks! It definitely looks like your correct. I suppose we never hit a bug with this because openocd always clears resumereq even though this is inconsisten with the prescribed W1 behaviour. See
https://github.com/riscv/riscv-openocd/blob/5cb2f200f828333b9daf5a2eeb11b0e26e635288/src/target/riscv/riscv-013.c#L3618. I think it makes sense to also raise this issue to `riscv-openocd`

Let me first run these changes through our tests and then I will report back.

bluewww, 2019-05-18:
I did some cleanup and merged in #18. Thanks for the report.

closed 2019-05-18 by bluewww