pulp-platform/riscv-dbg #23: Debug Version

zarubaf, 2019-06-22:
According to the latest Spec: `DbgVersion013` should be `1`. Not sure whether I've overseen this or they changed it.
https://github.com/pulp-platform/riscv-dbg/blob/6d768ac6371356b6e382c01ae7d860dd3e45248b/src/dm_pkg.sv#L20

imphil, 2019-06-25:
Are you sure you're looking at the right spec? I see this in the current master version:

https://github.com/riscv/riscv-debug-spec/blob/165f120a9eccc09343efd610761d1dfb50b06b2b/xml/dm_registers.xml#L100

>             1: There is a Debug Module and it conforms to version 0.11 of this
>            specification.
>
>            2: There is a Debug Module and it conforms to version 0.13 of this
>            specification.

zarubaf, 2019-06-25:
You are right, I was looking at the `sbversion` field. Where this is `1` as well as the `dtmcs` where it is `1` as well. Consistent naming for the win...

closed 2019-06-25 by zarubaf