pulp-platform/riscv-dbg #24: Unable to debug on virtual memory

niwis, 2019-06-28:
Hi,

for quite some time now I've been trying to debug s-mode code running on virtual memory  on Ariane using gdb and OpenOCD. As a first step, I'm attempting to read virtual memory using the program buffer as described in B.7.2 of the riscv debug specification. Therefore, I proceed as follows:

1. Create and install page tables, enable MMU in Sv39-mode
2. Enable s-mode. We're now running on virtual addresses
3. ebreak
4. make sure that mstatus.mpp equals 1 (s-mode)
5. set mstatus.mprv and dcrs.mprven

After the last instruction, the debug module becomes non-responsive. Apparently, the core is running in the park loop of the debug ROM. From there, it is instructed to signal the debug module that it's halted by a write to 0x100. However, since ld_st_translation is enabled, this write is diverted by the MMU.

From my point of view, this makes it impossible to operate the debug module with dcrs.mprven being set.

Are the methods and inferences above correct? Is that behavior something that is to be changed in the future? Do you have any experience in debugging lower privilege code on Ariane and can give me a hint on how to proceed?

Cheers and greetings,
Nils

zarubaf, 2019-06-28:
Hi Nils,

thanks for trying this. I think your approach makes sense. Looking at the specification I can't seem to find any more hints on how the hardware should handle this. 

1. My first feeling would be for the debugger to create a mapping on the zero page (`0x0` - `0x1000`). So it will allow the access to the debug ROM which is required to facilitate execution based debug. That should solve your problem.
2. I can't seem to find `PMA` attributes in the Privileged Specification which can exclude a memory region from address translation.
3. This is in an interesting problem for which I can't give you a certain answer, so it might be interesting to ask this on the debug spec GitHub Issue Tracker or RISC-V mailing list. It is certainly possible that there is something which we oversaw while implementing.

Best,
Florian

niwis, 2019-07-05:
Hi!

As you suggested, I‘ve consulted the riscv-mailing list (https://lists.riscv.org/g/tech-debug/message/833). There were several approaches suggested, one of which was to clear/restore `mprven` at the beginning/end of the debug rom. However, memory mapped I/O between the hart and the debug module seems to be required from outside the debug rom as well, for instance when accessing the abstract data registers during an abstract command. Hence, more sophisticated mechanisms (such as mapping abstract data or even the whole I/O to custom CSRs) are required to resolve this issue on HW level.

Returning to a more pragmatic level, I found another way to access virtual memory with the current implementation. The key is to set and clear `mprven` from within the program buffer right before and after the actual memory access. This way, the dm functionality remains unaffected. (https://github.com/riscv/riscv-openocd/pull/386)

Although I think this approach is valid, probably there are other and more efficient ones that should be supported as well. But I’ll leave that for you to decide.

Thanks for the support and hope this might help!

Nils

closed 2019-07-05 by niwis