pulp-platform/riscv-dbg #32: unsupported_command in dm_mem.sv

hyf6661669, 2019-09-19:
[image]

At here, we may have:
```
ac_ar.transfer = 0
```
And the document says:
[image]

If we have ```ac_ar.aarsize >= MaxAar & ac_ar.transfer == 0```, should we still make ```unsupported_command = 1``` ?

hyf6661669, 2019-09-19:
I think when ```ac_ar.transfer == 0``` and ```cmd_i.cmdtype == dm::AccessRegister```, ```unsupported_command``` should always be 0. In this situation, the abstract command will not be executed. So we don't need to care about whether ```aarsize, postexec, aarincrement, regno``` are valid or not.

bluewww, 2019-09-19:
I see. Then I think we need to change
```
end else if (ac_ar.aarsize >= MaxAar || ac_ar.aarpostincrement == 1'b1) begin
```
to
```
end else if ((ac_ar.aarsize >= MaxAar && ac_ar.transfer == 1)|| ac_ar.aarpostincrement == 1'b1) begin

```
and add your condition. Does this look right?

hyf6661669, 2019-09-20:
@bluewww 
LGTM.

closed 2020-04-29 by hyf6661669