pulp-platform/riscv-dbg #55: Number of scratch registers assertion race condition

jm4rtin, 2020-05-08:
The following assertion is inside an initial block:
```sv
// Fail if the DM is not located on the zero page and one hart doesn't have two scratch registers.
for (int i = 0; i < NrHarts; i++) begin
    assert ((DmBaseAddress > 0 && hartinfo_i[i].nscratch >= 2) || (DmBaseAddress == 0 && hartinfo_i[i].nscratch >= 1))
    else $fatal(1, "If the DM is not located at the zero page each hart needs at lest two scratch registers");
end
```

DmBaseAddress and NrHarts evaluate OK because they are parameters; however, hartinfo_i is a signal. Because hartinfo_i is a signal, there is a race condition in the simulation at 0 time between the assignment of the signal and the initial block above. In this case nscratch is read as X and the assertion fails. This race was observed with Xcelium.

A possible solution would be to add ```#0;``` prior to the loop, forcing this block to be evaluated last.

bluewww, 2020-05-08:
Thanks I see this is a race. Does `assert final` solve this for you?

jm4rtin, 2020-05-08:
I have never used ```assert final``` and so I am not 100% sure how it works but if the suggestion is to replace ```assert (``` with ```assert final (``` and no other changes, no it didn't work.

closed 2020-06-25 by bluewww