pulp-platform/riscv-dbg #57: Issue regarding master_be_o signal in dm_sba.sv / Wrong value during read transaction?

cr8601, 2020-05-11:
Hi,
within lowrisc/opentitan project I came to a problem when accessing the system bus via JTAG. Your documentation (https://github.com/pulp-platform/riscv-dbg/blob/master/doc/debug-system.md) states the following regarding this master_be signal:

be | 1 | output | Byte Enable. Is set for the bytes to write/read, sent together with req
-- | -- | -- | --

Within dm_sba.sv this signal is always 0 during read transactions. In my opinion this behaviour does not reflect the documentation... 

In case of opentitan this signal is used for the a_mask signal of the TLUL bus. I was suggested to raise this topic here in order to fix it at the root, if possible.

For my case simply inserting "be = '1;" into line 92 solved the problem.

Can someone please comment whether this could be changed here?

For reference here's the issue I raised within opentitan:
https://github.com/lowRISC/opentitan/issues/2126

Best regards
cr8601

bluewww, 2020-05-11:
It looks like the documentation is not entirely correct.
If you look here
https://github.com/pulp-platform/riscv-dbg/blob/3670267ab150f5a95e773bc18b1814e4031dc508/src/dm_csrs.sv#L527-L532

You see that currently we only fully support 32-bit or 64-bit accesses, though adding other access types doesn't seem too hard.

If you just hardcoded `be= '1` then you will break other things, namely openocd.

I see though that we have a real bug here: `sberror` is not signaled if someone tries to access with an access type that is not valid

cr8601, 2020-05-12:
Hi,
sorry, but I cannot really follow your comments. I think my problem is not related to any access type. I am talking about the host interface to the system bus which is defined by the ports master_*. There is a master_be port which is always tied to zero for read transactions (dm_sba.sv). The lowrisc/opentitan implementation (TLUL interface) uses these to mask the write and read bytes and therefore all system bus read requests, coming from external JTAG Debugger are failing (returning only zeros).

Can you please comment on this master_be signal? Is it always zero for read transactions by intention?

To simply get my system to work, I just put it to all 1 during read. Maybe this is not valid in general, but for now seems to fix my problem.

bluewww, 2020-05-12:
Sorry I misunderstood your problem description. Indeed you are correct, the be enable signal doesn't seem to be set correctly

zarubaf, 2020-05-12:
The `be` has no meaning in our system when performing a read (most of the stuff is AXI-based which doesn't use the `be` at all during reads) and is ignored. But seems an easy addition if that is necessary for the TL adapter.

cr8601, 2020-05-13:
It would be nice if this could be added!
Thanks
cr8601

Silabs-ArjanB, 2020-08-14:
Hi @cr8601 , @bluewww Hasn't this issue been solved now by the merge of https://github.com/pulp-platform/riscv-dbg/pull/70 ? If so, can it be closed?

closed 2020-08-14 by cr8601