pulp-platform/riscv-dbg #62: Operation status reading

omehed, 2020-06-02:
Hi, I am trying to understand how your debugger works, and am little bit confused about the behavior of the op and dmistat fields. Could you please comment on following situation.

There is some situation in which impossible to know on which operation DMI error flag is issued.
As an example, we have a very fast jtag and very slow system clock and perform next operations: 
> 1.	tdi: dmi write (op = 2) -> tdo: ok (op = 0)           // register write request
> 2.	tdi: dmi nop (op = 0) -> tdo: ok (op = 0) 	          // previous operation started succesfully
> 3.	tdi: dmi read (op = 1) -> tdo: busy (op = 3)       // register read request is not accepted
> 4.	tdi: reset error (dtmsc.dmistat = 1)                    // reset dmi error
> 5.	tdi: dmi read (op = 1) -> tdo: ok (op = 0)           // register read request  
> 6.	tdi: dmi nop (op = 0) -> tdo: busy (op = 3)        // read request is not accepted or in progress?

In situation described above write operation that started at stage 1 is not finished at stage 6 but at stage 5 we read status OK and we can assume that error read at stage 6 means that that reading at stage 5 is not finished but actually it even did not started and we should retry read request. Is this operation sequence not valid or debugger is not supports it? 
 Also reading error status by accessing dtmcs.dmistat returns status OK(0) but issues an error on debugger if current operation is not finished and next dtmcs.dmistat reading returns BUSSY(3) status, is this an correct behavior of debugger?

bluewww, 2020-06-11:
I believe the jtag clock always needs to be slower than the system clock in order for the dmi_jtag module to work, but I haven't looked deeper into it. Do you have a use case for such a fast jtag clock?

omehed, 2020-06-11:
Thank you for your response! No, I have no use case for such clock, It is just on simulation to understand capabilities of debug system

closed 2020-07-03 by bluewww