pulp-platform/riscv-dbg #63: abstract cmd error? or a typo?  when two dscratch implemented

gxflying, 2020-06-03:
Hi sir, I find there may be wrong in the code below:
1. the comment is  "store a0 in dscratch1" while used the "dm::csrr" which means that read a csr(dscratch1) and store it into a0.  so the code or comment maybe wrong here!
https://github.com/pulp-platform/riscv-dbg/blob/3670267ab150f5a95e773bc18b1814e4031dc508/src/dm_mem.sv#L357
2. the senario in which can execute a abstract cmd bellow is executing this sequence : "_entry" , "going", "whereto" and going into abstract cmd.  codes can be expanded bellow (assume dscratch1 implemented ):

_entry:
....
going:
#ifdef SND_SCRATCH
        sw zero, GOING(a0)          // When debug module sees this write, the GO flag is reset.
        **csrr a0, CSR_DSCRATCH1      // Restore a0 here**  **_where CSR_DSCRATCH1 contains the original value of a0_**
#else
        sw zero, GOING(zero)          // When debug module sees this write, the GO flag is reset.
#endif
        csrr s0, CSR_DSCRATCH0      // Restore s0 here
        jal zero, whereto

// store a0 in dscratch1 
dm::csrr(dm::CSR_DSCRATCH1, 5'd10)   **@AbstractCmdBaseAddr the first instruction **
here the values of CSR_DSCRATCH1 and a0 are the same (see the csrr a0, .. above)
so why should we move data between CSR_DSCRATCH1 & a0 when thay already the same?
so I think here no csr instruction is needed at all,   what do you think?
......

gxflying, 2020-06-08:
the csr write instructions stream below, which shows that the first instruction in the "abstract comand"  is WRONG, it should be a "dm::csr**w**(xx)".
the store & restore operation using csr should be a pair<dm::csrr, dm:csrw> where
dm::csrr : store data into csr from register
dm:csrw ：restore data from csr into register
https://github.com/pulp-platform/riscv-dbg/blob/3670267ab150f5a95e773bc18b1814e4031dc508/src/dm_mem.sv#L357
[image]

bluewww, 2020-06-08:
looks like you right

Silabs-ArjanB, 2020-10-12:
Hi @gxflying , @bluewww It seems to me that the same typo is present on the following line, do you agree with that?

https://github.com/pulp-platform/riscv-dbg/blob/f9cf9c40923151afe743fc06b2953731291eae17/src/dm_mem.sv#L398

gxflying, 2020-10-13:
@Silabs-ArjanB yeah， it seems that you are right, the code line 398 has the same bug

closed 2020-10-26 by bluewww