pulp-platform/riscv-dbg #65: OpenOCD and ndmreset signal reset issue

imi415, 2020-06-13:
Hi, I'm trying to integrate Ibex with debug module, however I found that under current implementation of the debug module, both reset-to-halt and reset-to-run are not working correctly, which crashes the OpenOCD process.

Below are my attempts to locate the problem, However, I have little knowledge about the code or the specification, if there are anything I got wrong or missed, please point out. Any suggestions are appreciated. Thanks in advance.

---
OpenOCD sets `NDMRESET` field in `dmcontrol` during reset period, and expects to see `ALLRUNNING` field to be set in dmstatus indicating the end of reset operation. These behaviors are addressed below:  
https://github.com/riscv/riscv-openocd/blob/4f9e2d7171f3cad8d1a99dff3eee5ec5e6d8ea2b/src/target/riscv/riscv-013.c#L2090

https://github.com/riscv/riscv-openocd/blob/4f9e2d7171f3cad8d1a99dff3eee5ec5e6d8ea2b/src/target/riscv/riscv-013.c#L2144

However, the debug module has its `ALLRUNNING` field wired to `halted` signal, which is set by a write operation to `HALT` address of the debug bus, and cleared by write to `RESUME` address of the debug bus.

https://github.com/pulp-platform/riscv-dbg/blob/3670267ab150f5a95e773bc18b1814e4031dc508/src/dm_csrs.sv#L249-L250

https://github.com/pulp-platform/riscv-dbg/blob/3670267ab150f5a95e773bc18b1814e4031dc508/src/dm_mem.sv#L242-L255

When the core is already in parked-loop state and a reset-to-run command is issued by OpenOCD, the core may never has a chance to write to `RESUME` address, since it has been resetted by `ndmreset` signal. So the `ALLRUNNING` field will never be set and OpenOCD just get stucked in the loop until timeout reached.

This internal state (core is actually running while dmstatus has wrong value) also causes crash of OpenOCD, then a reset to dm will be needed for the dm to function properly.

bluewww, 2020-06-13:
I guess they need to be put into the unavailable or running state after a ndmreset

gxflying, 2020-06-23:
yeah， this is because when ndmreset is set, the current version DM doesn't clear the "halted flags".

and one more thing, the dmcontrol.dmactive  may also not works well because it should do something with the registers in dm_mem.sv but it doesn't.
for example:
spec says dmactive can clear the abstractcs.busy status. but the current version does nothing with abstractcs.busy
[image]

bluewww, 2020-06-25:
Can you open a separate issue for this, please?

phsauter, 2026-08-07:
Closing as fixed: hart state is now cleared on ndmreset, and reset completion can be acknowledged through ndmreset_ack_i.

closed 2026-08-07 by phsauter