pulp-platform/riscv-dbg #74: Question about assertion in dm_top.sv

dawidzim, 2020-07-16:
Hey!
I have question about assertion in `src/dm_top.sv`:
``` 
     assert ((DmBaseAddress > 0 && hartinfo_i[i].nscratch >= 2) || (DmBaseAddress == 0 && hartinfo_i[i].nscratch >= 1))
        else $fatal(1, "If the DM is not located at the zero page each hart needs at lest two scratch registers %d %d",i, hartinfo_i[i].nscratch );
```
and in testbench `riscv_dbg/tb/tb_test_env` is:
```
 dm_top #(
       .NrHarts           ( NrHarts           ),
       .BusWidth          ( 32                ),
       .SelectableHarts   ( SELECTABLE_HARTS  )
    ) i_dm_top (

       .clk_i             ( clk_i             ),
       .rst_ni            ( rst_ni            ),
       .testmode_i        ( 1'b0              ),
       .ndmreset_o        ( ndmreset          ),
       .dmactive_o        (                   ), // active debug session TODO
       .debug_req_o       ( dm_debug_req      ),
       .unavailable_i     ( ~SELECTABLE_HARTS ),
       .hartinfo_i        ( '0                ),
```
so `hartinfo_i` is always 0 here, so this assert will call $fatal.

bluewww, 2020-07-16:
Setting `hartinfo_i` to `0` was just me being lazy.

bluewww, 2020-10-26:
Fixed in #82

closed 2020-10-26 by bluewww