pulp-platform/riscv-dbg #76: Requirements on clk_i in relationship to tck_i and related constraints

Silabs-ArjanB, 2020-07-29:
The riscv-dbg contains a clock domain crossing implemented inside dmi_cdc.sv.

clk_i and tck_i are allowed to be asynchronous to each other (please correct me if I am wrong about this). It is not clear however if there are any relative frequency requirements, e.g. clk_i must be at least N times faster than tck_i. Could you please indicate if there are any relative frequency requirements for riscv-dbg to work? Is it okay for either of the clocks to be completely stopped and if so, can that lead to continuously stalled transactions on the other interface?

Similarly are the two resets completely independent or are there any assumptions on when they are allowed to be asserted/deasserted relative to each other?

I see that cdc_2phase.sv contains the following statement:

```
> /// CONSTRAINT: Requires max_delay of min_period(src_clk_i, dst_clk_i) through
> /// the paths async_req, async_ack, async_data.
```
Does this imply that we need a synthesis constraint to properly get riscv-dbg to work and if so do you have an example of such sdc constraint? Or does the above remark boil down to a relative frequency requirement?

noytzach, 2020-09-10:
See #72 
It fixes an issue where clk is slower (or not much faster) than tck.

Silabs-ArjanB, 2020-10-06:
Hi @noytzach Thanks!