pulp-platform/riscv-dbg #80: Vivado/ModelSim simulation Hanging

adeel47, 2020-09-15:
I am using riscv-dbg IP with RI5CY core in my design which is running at 25MHz. I am simulating my design Using Vivado (Same issue in ModelSim as well). In Vivado when testing the riscv-dbg tests, in file dm_mem.sv, at line 239 when we got a new request which is a write and the address addr_i becomes HaltedAddr, the Vivado simulation hangs there and is not proceeding anymore, which seems like a race condition has been started. While further debugging the issue, it seems that the simulation gets in a loop between read/write logic always_comb block (line 239) and hart ctrl queue always_comb block (line 136). 
I found a temporarily solution as when i replace the always_comb block of hart ctrl queue logic with always@(posedge clk_i), the issue is gone. May i please know what is happening and what could be the right solution of the problem?

bluewww, 2020-10-01:
So basically two always_comb block are triggering each other forever? I can't reproduce this bug

adeel47, 2020-10-01:
Yes, both the always_comb blocks are triggering forever. Which simulator are you using?

bluewww, 2020-10-01:
Questa Sim-64 vsim 10.7b_1 Simulator 2018.07 Jul 26 2018
or
verilator

majbthrd, 2020-12-13:
I was seeing the same thing in Vivado v2020.2 and have submitted pull request #102 with a suggested fix.