pulp-platform/riscv-dbg #81: sba bursts in verilator simulation

joestoy, 2020-09-30:
This concerns commit f9cf9c4.  The attached screenshot, of a verilator simulation, shows a dmi request for an sba burst: the sbcs word, the address, and then the sequence of data words. Underneath is the debug module's master traffic.  Note that address 0xC0000004 gets the data which in the dmi request corresponds to 0xC0000018.  The intervening data items have been dropped.  (No doubt this problem would not occur with an FPGA emulation, where the dut would be running relatively much faster.)

Would it be possible to defer asserting ```dmi_resp_valid_o``` until ```master_r_valid_i``` has been asserted?

[image]

noytzach, 2020-10-01:
I think the correct behavior according to the spec should be:
* First write to `sbdata` - initiate bus access
* Following write to `sbdata` (while previous access is not over, as in waves) should set `sbbusyerror`.
* Following writes to `sbdata` should do nothing until `sbbusyerror` is cleared by writing 1.

### from 3.12.23 `sbdata`
> If the bus master is busy then accesses set `sbbusyerror`, and don’t do anything else.
and
> If `sberror` or `sbbusyerror` both aren’t 0 then accesses do nothing

joestoy, 2020-10-01:
Yes, I accept that.  So the second write should set sbbusyerror and do nothing else; in particular, the master write to 0xC0000004 (with the incorrect data) shouldn't happen, and at the end of the burst the address register should contain 0xC0000004.  In fact it contains 0xC0000008.

This next part is tentative: I'd like to see what happens after the bug described above is fixed.  But what seems to happen now is this.  openocd is supposed to react by trying again with a longer interval between beats.  In fact it first does a non-incrementing write to 0xC0000008, which is successful, and then tries a single incrementing write to the same address.  But the debug module appears not to increment the address as it should; so openocd goes on writing to 0xC0000008 forever.

joestoy, 2020-10-08:
@zarubaf @noytzach:  May I respectfully suggest some amendments to ```dm_csrs.sv```, as shown in the "git diff" below (which refers to commit f9cf9c4 of the repo)?

All but the first change are to ensure that no transactions may be initiated if sbbusyerror is set, thus inhibiting the erroneous second master transaction in the screenshot at the top of this thread.

The first change is a workaround for another error, which I also tentatively alluded to in an earlier post.  The symptom is that once an sbbusyerror has occurred, autoincrementing is turned off.  The sbbusyerror bit in sbcs is R/W1C.  The openocd we are using (```Open On-Chip Debugger 0.10.0+dev-g27c0fd7a (2020-01-04-05:57)```) clears this bit by writing 0x00400000 to sbcs; this does clear sbbusyerror, but it also clears all the R/W bits in the word, including autoincrement.  openocd's subsequent activity clearly expects this not to happen. This looks like an openocd error; but until it is fixed, the first change in the diff accommodates this behaviour.

Please feel free, of course, to make any changes to these amendments as you wish.

```
diff --git a/src/dm_csrs.sv b/src/dm_csrs.sv
index 78fd32a..ac9257f 100644
--- a/src/dm_csrs.sv
+++ b/src/dm_csrs.sv
@@ -434,15 +434,19 @@ module dm_csrs #(
             sbcs_d.sbbusyerror = 1'b1;
           end else begin
             sbcs = dm::sbcs_t'(dmi_req_i.data);
-            sbcs_d = sbcs;
-            // R/W1C
-            sbcs_d.sbbusyerror = sbcs_q.sbbusyerror & (~sbcs.sbbusyerror);
-            sbcs_d.sberror     = sbcs_q.sberror     & (~sbcs.sberror);
+           // workaround for openocd behavior:
+           if (sbcs.sberror | sbcs.sbbusyerror) begin
+               // R/W1C
+               sbcs_d.sbbusyerror = sbcs_q.sbbusyerror & (~sbcs.sbbusyerror);
+               sbcs_d.sberror     = sbcs_q.sberror     & (~sbcs.sberror);
+           end else begin
+               sbcs_d = sbcs;
+           end
           end
         end
         dm::SBAddress0: begin
           // access while the SBA was busy
-          if (sbbusy_i) begin
+          if (sbbusy_i | sbcs_q.sbbusyerror) begin
             sbcs_d.sbbusyerror = 1'b1;
           end else begin
             sbaddr_d[31:0] = dmi_req_i.data;
@@ -451,7 +455,7 @@ module dm_csrs #(
         end
         dm::SBAddress1: begin
           // access while the SBA was busy
-          if (sbbusy_i) begin
+          if (sbbusy_i | sbcs_q.sbbusyerror) begin
             sbcs_d.sbbusyerror = 1'b1;
           end else begin
             sbaddr_d[63:32] = dmi_req_i.data;
@@ -459,7 +463,7 @@ module dm_csrs #(
         end
         dm::SBData0: begin
           // access while the SBA was busy
-          if (sbbusy_i) begin
+          if (sbbusy_i | sbcs_q.sbbusyerror) begin
            sbcs_d.sbbusyerror = 1'b1;
           end else begin
             sbdata_d[31:0] = dmi_req_i.data;
@@ -468,7 +472,7 @@ module dm_csrs #(
         end
         dm::SBData1: begin
           // access while the SBA was busy
-          if (sbbusy_i) begin
+          if (sbbusy_i | sbcs_q.sbbusyerror) begin
            sbcs_d.sbbusyerror = 1'b1;
           end else begin
             sbdata_d[63:32] = dmi_req_i.data;
```

Silabs-ArjanB, 2020-10-12:
Hi @joestoy, the first fix (related to writing 0x00400000 to sbcs) is indeed a work-around; I think it is better to file a bug/issue on OpenOCD for that as the work-around is very specific to OpenOCD and might break other debuggers.

joestoy, 2020-10-13:
@Silabs-ArjanB: agreed.  We are submitting a bug issue to openocd, and will be using the workaround only until they fix it.  (The other fixes, though, are of course for riscv-dbg itself.)

closed 2020-10-26 by bluewww

jm4rtin, 2021-03-08:
@joestoy Can you please provide the link to the OpenOCD issue that was created?

joestoy, 2021-03-08:
@jm4rtin: See https://github.com/riscv/riscv-openocd/pull/547#issue-503198816