pulp-platform/riscv-dbg #96: Giving an error for ABI in testbench

ShraddhaDevaiya, 2020-10-30:
Hello, I am trying to run testbench code(from /tb) of riscv-dbg. It is giving an error like following:
```
riscv64-unknown-elf-gcc/bin/riscv64-unknown-elf-gcc -march=rv32imc -Os -g  -nostdlib -static -T prog/link.ld prog/test.c prog/crt0.S prog/syscalls.c prog/vectors.S -lc -lm -lgcc -o prog/test.elf
cc1: error: requested ABI requires -march to subsume the 'D' extension
cc1: error: ABI requires -march=rv64
cc1: error: requested ABI requires -march to subsume the 'D' extension
cc1: error: ABI requires -march=rv64
cc1: error: requested ABI requires -march to subsume the 'D' extension
cc1: error: ABI requires -march=rv64
cc1: error: requested ABI requires -march to subsume the 'D' extension
cc1: error: ABI requires -march=rv64
make: *** [Makefile:283: prog/test.elf] Error 1
```
Can anyone please help to resolve this ?

bluewww, 2020-10-30:
You built your compiler with the float abi (ilp32f or ilp64f). This means you need to tell the compiler to also emit float insn. So put march=rv32imfc.

closed 2020-10-30 by bluewww

ShraddhaDevaiya, 2020-10-31:
Hi @bluewww , I have tried this, but still it was giving the same error like following : 
```
riscv64-unknown-elf-gcc/bin/riscv64-unknown-elf-gcc -march=rv32imfc -Os -g  -nostdlib -static -T prog/link.ld prog/test.c prog/crt0.S prog/syscalls.c prog/vectors.S -lc -lm -lgcc -o prog/test.elf
cc1: error: requested ABI requires -march to subsume the 'D' extension
cc1: error: ABI requires -march=rv64
cc1: error: requested ABI requires -march to subsume the 'D' extension
cc1: error: ABI requires -march=rv64
cc1: error: requested ABI requires -march to subsume the 'D' extension
cc1: error: ABI requires -march=rv64
cc1: error: requested ABI requires -march to subsume the 'D' extension
cc1: error: ABI requires -march=rv64
make: *** [Makefile:283: prog/test.elf] Error 1
```
What can I do next?

bluewww, 2020-10-31:
probably `imdc` instead of `imfc`. I mean look it says it right in the error message!

ShraddhaDevaiya, 2020-11-03:
Hi @bluewww , thanks for helping. it is working now.