pulp-platform/riscv-dbg #97: behavior of testbench

ShraddhaDevaiya, 2020-11-03:
Hello,
I am trying to run testbench of riscv debug module. For that, I have run command ```make veri-run```. For this it is just stopped at following flow:

[image]

Is this in interactive mode, or all configurations and commands are  passed in **pulpissimo_debug.cfg** file ?
Can anyone please put some light on this?

bluewww, 2020-11-03:
Now you can connect to it with openocd using one of the `*.cfg` openocd scripts

ShraddhaDevaiya, 2020-11-04:
Hi @bluewww , I am trying to run this command ```openocd -d 1 -f dm_debug.cfg``` after this prompt, but it is giving an output like this : 
```
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
User : 13 2 command.c:544 command_print(): debug_level: 3
Debug: 14 2 options.c:181 add_default_dirs(): bindir=/usr/bin
Debug: 15 2 options.c:182 add_default_dirs(): pkgdatadir=/usr/share/openocd
Debug: 16 2 options.c:183 add_default_dirs(): exepath=/usr/bin
Debug: 17 2 options.c:184 add_default_dirs(): bin2data=../share/openocd
Debug: 18 2 configuration.c:42 add_script_search_dir(): adding /home/adduser/.openocd
Debug: 19 2 configuration.c:42 add_script_search_dir(): adding /usr/bin/../share/openocd/site
Debug: 20 2 configuration.c:42 add_script_search_dir(): adding /usr/bin/../share/openocd/scripts
Debug: 21 6 configuration.c:82 find_file(): found dm_debug.cfg
Debug: 22 11 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_debug_level 4
Debug: 23 11 command.c:143 script_debug(): command - debug_level ocd_debug_level 4
Error: 25 11 log.c:238 handle_debug_level_command(): level must be between -3 and 3
Debug: 26 11 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_usage debug_level
Debug: 27 11 command.c:143 script_debug(): command - usage ocd_usage debug_level
User : 29 11 command.c:865 command_help_show(): debug_levelUser : 30 11 command.c:871 command_help_show():  User 
: 31 11 command.c:844 command_help_show_wrap(): number
User : 32 11 command.c:687 command_run_line():
```
Is this expected or I am missing something. can you please help?

bluewww, 2020-11-04:
Debug: 23 11 command.c:143 script_debug(): command - debug_level ocd_debug_level 4
Error: 25 11 log.c:238 handle_debug_level_command(): level must be between -3 and 3

We set the debug level to 4 in the openocd script. Maybe openocd changed the allowable ranges recentl,

ShraddhaDevaiya, 2020-11-05:
Hi @bluewww,  I have changed the debug level to 2 and then run the script. Then it is giving an some other error like following:
```
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
User : 13 2 command.c:544 command_print(): debug_level: 3
Debug: 14 2 options.c:181 add_default_dirs(): bindir=/usr/bin
Debug: 15 2 options.c:182 add_default_dirs(): pkgdatadir=/usr/share/openocd
Debug: 16 2 options.c:183 add_default_dirs(): exepath=/usr/bin
Debug: 17 2 options.c:184 add_default_dirs(): bin2data=../share/openocd
Debug: 18 2 configuration.c:42 add_script_search_dir(): adding /home/adduser/.openocd
Debug: 19 2 configuration.c:42 add_script_search_dir(): adding /usr/bin/../share/openocd/site
Debug: 20 2 configuration.c:42 add_script_search_dir(): adding /usr/bin/../share/openocd/scripts
Debug: 21 4 configuration.c:82 find_file(): found dm_debug.cfg
Debug: 22 8 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_debug_level 2
Debug: 23 8 command.c:143 script_debug(): command - debug_level ocd_debug_level 2
debug_level: 2
adapter speed: 10000 kHz
Warn : Adapter driver 'remote_bitbang' did not declare which transports it allows; assuming legacy JTAG-only
Info : only one transport option; autoselect 'jtag'
dm_debug.cfg:7: Error: can't read "::env(JTAG_VPI_PORT)": no such element in array
in procedure 'script'
at file "embedded:startup.tcl", line 60
at file "dm_debug.cfg", line 7
```
How can I resolve this?

noytzach, 2020-11-05:
Hi @ShraddhaDevaiya,

See example in [README.md](https://github.com/pulp-platform/riscv-dbg/blob/master/tb/README.md). You should run the following command *before* running openocd:
```
export JTAG_VPI_PORT=<port_name_from 1>
```

If using csh/tcsh, you should use this command instead:
```
setenv JTAG_VPI_PORT <port_name_from 1>
```

ShraddhaDevaiya, 2020-11-06:
Hi @noytzach , I did the same as per your instructions. But still it is showing some error:
```
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2   
For bug reports, read       
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 2
adapter speed: 10000 kHz
Warn : Adapter driver 'remote_bitbang' did not declare which transports it allows; assuming legacy JTAG-onlyInfo : only one transport option; autoselect 'jtag'
TAP: riscv.cpu

embedded:startup.tcl:21: Error: Unknown target type riscv, try one of arm7tdmi, arm9tdmi, arm920t, arm720t, 
arm966e, arm946e, arm926ejs, fa526, feroceon, dragonite, xscale, cortex_m, cortex_a, cortex_r4, arm11, ls1_sap, mips_m4k, avr, dsp563xx, dsp5680xx, testee, avr32_ap7k, hla_target, nds32_v2, nds32_v3, nds32_v3m, or1k, quark_x10xx,  or quark_d20xx
in procedure 'script'
at file "embedded:startup.tcl", line 60
in procedure 'target' called at file "dm_compliance_test.cfg", line 18
in procedure 'ocd_bouncer'
at file "embedded:startup.tcl", line 21
```
I think riscv target is unrecongnized, do I need to install some dependencies for this?
is this expected, or I am missing something?

noytzach, 2020-11-06:
You have to install [riscv-openocd](https://www.github.com/riscv/riscv-openocd)

And run this one (using fullpath) and not the one you already have, which does not support riscv.

ShraddhaDevaiya, 2020-11-10:
Hi, Sorry for the delay.
I have installed openocd which has riscv support, and then I tried to run same command to run file **dm_compliance_test.cfg** . Now I am getting following error:
```
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
User : 13 13 command.c:544 command_print(): debug_level: 3
Debug: 14 13 options.c:181 add_default_dirs(): bindir=/usr/bin
Debug: 15 13 options.c:182 add_default_dirs(): pkgdatadir=/usr/share/openocd
Debug: 16 13 options.c:183 add_default_dirs(): exepath=/usr/bin
Debug: 17 13 options.c:184 add_default_dirs(): bin2data=../share/openocd
Debug: 18 13 configuration.c:42 add_script_search_dir(): adding /home/adduser/.openocd
Debug: 19 13 configuration.c:42 add_script_search_dir(): adding /usr/bin/../share/openocd/site
Debug: 20 13 configuration.c:42 add_script_search_dir(): adding /usr/bin/../share/openocd/scripts
User : 21 35 command.c:687 command_run_line(): embedded:startup.tcl:60: Error: Can't find openocd.cfg
in procedure 'script'
at file "embedded:startup.tcl", line 60
Debug: 22 35 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_init
Debug: 23 35 command.c:143 script_debug(): command - init ocd_init
Debug: 25 35 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_target init
Debug: 26 35 command.c:143 script_debug(): command - ocd_target ocd_target init
Debug: 28 35 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_target names        
Debug: 29 35 command.c:143 script_debug(): command - ocd_target ocd_target names
Debug: 30 35 target.c:1324 handle_target_init_command(): Initializing targets...
Error: 31 35 core.c:1332 adapter_init(): Debug Adapter has to be specified, see "interface" command       
Debug: 32 35 command.c:626 run_command(): Command failed with error code -101
User : 33 35 command.c:687 command_run_line(): embedded:startup.tcl:60: Error:
in procedure 'script'
at file "embedded:startup.tcl", line 60
```
Can you please help me?

noytzach, 2020-11-11:
It looks like you did not use the -f option (as in the README) so openocd is looking for the default script (openocd.cfg).

```
/path/to/riscv-openocd/bin/openocd -f dm_compliance_test.cfg
```

ShraddhaDevaiya, 2020-11-13:
Yeah got that. I am facing issue to install riscv target in openocd. Let me complete that, then will check again.

gzhan88, 2020-11-20:
@ShraddhaDevaiya 
Hello! Do you solve all the problems?
Below is my new issue:
environment:
1.cenos7
2.bashrc set: 
(1)export RISCV=/opt/tools/riscv/toolchain
    export PATH=$PATH:$RISCV/bin

(2)export OPENOCD_HOME=/opt/tools/riscv-openocd
    export PATH=$PATH:$OPENOCD_HOME/bin

Process:
**1.one Terminal**
[gzl@gzl tb]$ make veri-run
verilator --cc --sv --exe  \
	--Wno-lint --Wno-UNOPTFLAT --Wno-BLKANDNBLK \
	--Wno-MODDUP +incdir+riscv/rtl/include --top-module \
	tb_top_verilator --Mdir cobj_dir \
	-CFLAGS "-std=gnu++11 -O2"  \
	fpnew/src/fpnew_pkg.sv dm_tb_pkg.sv riscv/rtl/include/cv32e40p_apu_core_pkg.sv riscv/rtl/include/cv32e40p_pkg.sv riscv/rtl/include/../../bhv/include/cv32e40p_tracer_pkg.sv ../src/dm_pkg.sv boot_rom.sv dp_ram.sv mm_ram.sv SimJTAG.sv tb_top_verilator.sv riscv/rtl/../bhv/cv32e40p_sim_clock_gate.sv riscv/rtl/../bhv/cv32e40p_tracer.sv riscv/rtl/cv32e40p_if_stage.sv riscv/rtl/cv32e40p_cs_registers.sv riscv/rtl/cv32e40p_register_file_ff.sv riscv/rtl/cv32e40p_load_store_unit.sv riscv/rtl/cv32e40p_id_stage.sv riscv/rtl/cv32e40p_aligner.sv riscv/rtl/cv32e40p_decoder.sv riscv/rtl/cv32e40p_compressed_decoder.sv riscv/rtl/cv32e40p_fifo.sv riscv/rtl/cv32e40p_prefetch_buffer.sv riscv/rtl/cv32e40p_hwloop_regs.sv riscv/rtl/cv32e40p_mult.sv riscv/rtl/cv32e40p_int_controller.sv riscv/rtl/cv32e40p_ex_stage.sv riscv/rtl/cv32e40p_alu_div.sv riscv/rtl/cv32e40p_alu.sv riscv/rtl/cv32e40p_ff_one.sv riscv/rtl/cv32e40p_popcnt.sv riscv/rtl/cv32e40p_apu_disp.sv riscv/rtl/cv32e40p_controller.sv riscv/rtl/cv32e40p_obi_interface.sv riscv/rtl/cv32e40p_prefetch_controller.sv riscv/rtl/cv32e40p_sleep_unit.sv riscv/rtl/cv32e40p_core.sv common_cells/src/cdc_2phase.sv common_cells/src/fifo_v2.sv common_cells/src/fifo_v3.sv common_cells/src/rstgen.sv common_cells/src/rstgen_bypass.sv tech_cells_generic/src/cluster_clock_inverter.sv tech_cells_generic/src/pulp_clock_mux2.sv tech_cells_generic/src/cluster_clock_gating.sv ../debug_rom/debug_rom.sv ../src/dm_csrs.sv ../src/dmi_cdc.sv ../src/dmi_jtag.sv ../src/dmi_jtag_tap.sv ../src/dm_mem.sv ../src/dm_sba.sv ../src/dm_top.sv ../src/dm_obi_top.sv \
	-LDFLAGS "-L../remote_bitbang \
	-Wl,--enable-new-dtags -Wl,-rpath,remote_bitbang -lrbs_veri" \
	tb_top_verilator.cpp
cd cobj_dir && make -f Vtb_top_verilator.mk
make[1]: Entering directory `/home/gzl/20201120/riscv-dbg/tb/cobj_dir'
g++  -I.  -MMD -I/opt/tools/verilator/share/verilator/include -I/opt/tools/verilator/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -std=gnu++11 -O2  -std=gnu++11 -Os -c -o tb_top_verilator.o ../tb_top_verilator.cpp
g++  -I.  -MMD -I/opt/tools/verilator/share/verilator/include -I/opt/tools/verilator/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -std=gnu++11 -O2  -std=gnu++11 -Os -c -o verilated.o /opt/tools/verilator/share/verilator/include/verilated.cpp
g++  -I.  -MMD -I/opt/tools/verilator/share/verilator/include -I/opt/tools/verilator/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -std=gnu++11 -O2  -std=gnu++11 -Os -c -o verilated_dpi.o /opt/tools/verilator/share/verilator/include/verilated_dpi.cpp
g++  -I.  -MMD -I/opt/tools/verilator/share/verilator/include -I/opt/tools/verilator/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -std=gnu++11 -O2  -std=gnu++11 -Os -c -o Vtb_top_verilator.o Vtb_top_verilator.cpp
g++  -I.  -MMD -I/opt/tools/verilator/share/verilator/include -I/opt/tools/verilator/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -std=gnu++11 -O2  -std=gnu++11 -Os -c -o Vtb_top_verilator__1.o Vtb_top_verilator__1.cpp
g++  -I.  -MMD -I/opt/tools/verilator/share/verilator/include -I/opt/tools/verilator/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -std=gnu++11 -O2  -std=gnu++11 -Os -c -o Vtb_top_verilator___024unit.o Vtb_top_verilator___024unit.cpp
g++  -I.  -MMD -I/opt/tools/verilator/share/verilator/include -I/opt/tools/verilator/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -std=gnu++11 -O2  -std=gnu++11 -Os -c -o Vtb_top_verilator__Dpi.o Vtb_top_verilator__Dpi.cpp
g++  -I.  -MMD -I/opt/tools/verilator/share/verilator/include -I/opt/tools/verilator/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -std=gnu++11 -O2  -std=gnu++11  -c -o Vtb_top_verilator__Slow.o Vtb_top_verilator__Slow. [... cut ...]
Error: 73 1 riscv_debug.c:39 riscv_create_rtos(): target. See
Error: 74 1 riscv_debug.c:40 riscv_create_rtos(): https://github.com/riscv/riscv-tests/blob/ec6537fc4a527ca88be2f045e01c460e640ab9c5/debug/targets/SiFive/HiFiveUnleashed.cfg#L11
Error: 75 1 riscv_debug.c:41 riscv_create_rtos(): for an example.
User : 76 1 command.c:692 command_run_line(): 
[gzl@gzl tb]$ 

#############################################################
I don't know how to solve the  problem.@ShraddhaDevaiya  @noytzach,do you any ideas? Thank you very much!

Lucas-Wye, 2024-10-16:
I also encountered the same problem. Which version of openocd should we used?

@ShraddhaDevaiya @noytzach @bluewww @gz2023 @rswarbrick

bluewww, 2024-10-16:
Try this version for running the tests https://github.com/pulp-platform/riscv-openocd

The problem is that upstream removed the built-in tests at some point so the whole setup is a bit obsolete, but atm it is the best we've got (until I find some time to clean up this mess)

Lucas-Wye, 2024-10-16:
> Try this version for running the tests https://github.com/pulp-platform/riscv-openocd
> 
> The problem is that upstream removed the built-in tests at some point so the whole setup is a bit obsolete, but atm it is the best we've got (until I find some time to clean up this mess).

what is the commit id of openocd that was used to successfully setup?

phsauter, 2026-08-07:
Closing as obsolete because the required OpenOCD revision and complete compliance invocation are now pinned and automated under ci.