pulp-platform/riscv-dbg #113: TestLogicReset dose not work. The value of idcode_q/bypass_d did not switch back to default value when TestLogicReset

sywang33, 2021-06-25:
Issue 1 :
At line 219 In file dmi_jtag_tap.sv, the trst_no signals assigned to 1'b1.  It should be assign to 1'b0, isn't it? 
Here is the code:
unique case (tap_state_q)
      TestLogicReset: begin
        tap_state_d = (tms_i) ? TestLogicReset : RunTestIdle;
          trst_no = 1'b1;   // <------  should be 1'b0 ??
      end

Issue 2:
Should below signals switch back to default value when TAP in  TestLogicReset  state.
These signals include:
  1. idcode_q
  2. bypass_q
  3. jtag_ir_shift_q
  4. jtag_ir_q

Previous version,  these signals change back to default value when   TestLogicReset  state

Robin

andreaskurth, 2022-09-27:
## On Issue 1
`dmi_jtag_tap` no longer has an (asynchronous) `trst_no` output; the functional/"warm" reset mechanism has been changed to a synchronous signal in #123. The polarity matches the signal name:
https://github.com/pulp-platform/riscv-dbg/blob/ec0d92fa5fdaaf611e5b8a4cc8ee84017ff98710/src/dmi_jtag_tap.sv#L223-L225
So I think Issue 1 is solved.

## On Issue 2

The `dmi_jtag_tap` module now sets `test_logic_reset` and `dmi_clear_o` in the `TestLogicReset` state, which clears `jtag_ir_d` to `IDCODE` (and thus on the next posedge of `tck_i` also `jtag_ir_q`).

The other registers, `idcode_q`, `bypass_q`, and `jtag_ir_shift_q`, are currently *not* cleared in the `TestLogicReset` state.

I think it should clear them, though: The JTAG spec (IEEE 1149.1-2013) defines the `trst_n` input to be optional, so `TestLogicReset` is a canonical way to bring the TAP to the known initial state. (I did not find this explicitly in the spec, however.) I'll open a PR to implement this clearing.

bluewww, 2022-10-04:
Fixed in #139