pulp-platform/riscv-dbg #121: Software Breakpoint

sriharich20, 2021-12-24:
Can some one explain how ebreak instruction will be replaced in the breakpoint address by GDB. Wheter ebreak is written to breakpoint address by using system bus access (SBA) or using program buffer?

bluewww, 2021-12-24:
I think this depends on the software stack you are using. Both ways are possible.

closed 2021-12-26 by bluewww

sriharich20, 2022-01-03:
In this implementation how ebreak is replaced at software breakpoints

sriharich20, 2022-01-04:
> I think this depends on the debug software you are using. Both ways are possible.

if the ebreak instruction is replaced by debugger configuring the program buffer then hart has to execute the program buffer which contains the instruction writing ebreak to the instruction memory is that possible