pulp-platform/riscv-dbg #131: compile error: can't find module cdc_2phase_clearable

damnsavage, 2022-04-27:
Hi,
I just downloaded this repo it looks interesting thanks for creating it!
However I have a problem when trying to run it:

> cd riscv-dbg/tb
> make veri-run
> %Error: ../src/dmi_cdc.sv:50: Cannot find file containing module: **'cdc_2phase_clearable'**
>  cdc_2phase_clearable #(.T(dm::dmi_req_t)) i_cdc_req (

I see this module was added in February 2022 but the Makefile checks out a common_cells version from 2019
> git checkout $(COMMON_SHA)

If I comment this line in the Makefile and use the latest common_cells, then change cdc_2phase to cdc_2phase_clearable  I get another issue:
> make: *** No rule to make target 'common_cells/src/fifo_v2.sv', needed by 'testbench_verilator'.  Stop.

fifo_v2 was deprecated in the latest common_cells. So I copied it back into the common_cells/src directory.

Then I get additional compile errors which makes me think I'm doing something drastically wrong...
I tried pulling all the latest versions of the tb dependent repos and commented the versions in the Makefile 
> CV32E40P_SHA       = f9d63290eea738cb0a6fbf1e77bbd18555015a03
> FPU_SHA                 = v0.6.1
> COMMON_SHA      = 337f54a7cdfdad78b124cbdd2a627db3e0939141
> TECH_SHA               = b35652608124b7ea813818b14a00ca76edd7599d

Then I get this error
> make: *** No rule to make target 'tech_cells_generic/src/cluster_clock_inverter.sv', needed by 'testbench_verilator'.  Stop.

I'm on version 69be5ddc03ea1688c0eab47d6ed9d0e8725beda1 of riscv-dbg.
Can you help me? Should I use one of the annotated tag versions?

bluewww, 2022-04-27:
The dependencies are listed in the `Bender.yml` file. Looks like we forgot to update the testbench dependencies which are manually introduced.

damnsavage, 2022-06-08:
Hi, 
Thanks for your reply. I played a little with bender, e.g. ./bender update, bender checkout etc... but I don't think it did anything. 

Reading Bender.yml, I manually checked out the relevant common_cells and tech_cells_generic versions.
However I still have compile errors for example can't find common_cells/src/fifo_v2.sv,I see only common_cells/src/fifo_v3.sv.
I went back to a much older version of common_cells than then one in Bender.yml, I took this one from 3yrs ago (0787d3e) which has fifo_v2.sv but after I've errors from the tech_cells_generic missing files (tech_cells_generic/src/cluster_clock_inverter.sv) which I didn't find in any release.

Would it be possible to make another release tag?

Best regards

Damian

bluewww, 2022-06-08:
If you make a pr with the fixes in the bender.yml I can do that