pulp-platform/riscv-dbg #137: Assertion failure expected?

dpetrisko, 2022-09-20:
Hi, I'm trying to use the pulp debug module for my own chip. Things seems to be mainly working, but I'm trying to sort out the last few bugs. I'm using a custom DMI/DTM, but standard riscv-openocd, PULP DM and debug_rom.

Openocd version:
```
Open On-Chip Debugger 0.11.0+dev-02377-g911d68ef2 (2022-09-12-11:22)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
```

testchip.cfg
```
adapter driver remote_bitbang
remote_bitbang host localhost
remote_bitbang port 12999
transport select jtag

set _CHIPNAME testchip
set _CPUTAPID 0x04f5484d

jtag newtap $_CHIPNAME cpu -irlen 5 -expected-id $_CPUTAPID
set _TARGETNAME $_CHIPNAME.cpu
target create $_TARGETNAME riscv -chain-position $_TARGETNAME -coreid 0

riscv set_reset_timeout_sec 2000
riscv set_command_timeout_sec 2000

# prefer to use sba for system bus access
riscv set_prefer_sba off

init
halt
```

I added a print debug to see what DTM requests I'm receiving and found this violated assertion.
```
...
[          1516600000] '{addr:'h5, op:DTM_READ, data:'h0}
"PATH/opentitan/hw/ip/prim/rtl/prim_fifo_sync.sv", 193: testbench.dm.i_dm_top.i_dm_csrs.i_fifo.DataKnown_A: started at 1516601500ps failed at 1516601500ps
    Offending '(!$isunknown(rdata_o))'
Error: "PATH/opentitan/hw/ip/prim/rtl/prim_fifo_sync.sv", 193: testbench.dm.i_dm_top.i_dm_csrs.i_fifo.DataKnown_A: at time 1516601500 ps
1516602000: (PATH/opentitan/hw/ip/prim/rtl/prim_fifo_sync.sv:193) [testbench.dm.i_dm_top.i_dm_csrs.i_fifo.DataKnown_A] [ASSERT FAILED] DataKnown_A

```

Looking back through my log, there is no DTM_WRITE to addr 5 before this read which explains the Xs. However, I was wondering if it were perhaps a software/driver problem that these writes were not happening or getting dropped. Have you seen this problem before?

Thanks!

bluewww, 2022-09-21:
I haven't seen this kind of error before.

You seem to be using a very recent openocd version and a different FIFO IP it seems. The FIFO IP that is supported by riscv-dbg is the one from [common_cells](https://github.com/pulp-platform/common_cells/).

dpetrisko, 2022-09-21:
Oh, I hadn't noticed that opentitan patches the pulp DM with different primitives, so it's not standard. My mistake!  

Which version of Openocd do you suggest? I just pulled down latest and built