pulp-platform/riscv-dbg #149: questions about -rtos riscv in dm_compliance_test.cfg

coderalai, 2022-12-16:
hi!
After I have finished the compilation of rtl and prog/test.c by using vcs and gcc, i run the command "`//opt/openocd/bin/openocd -f dm_compliance_test.cfg`". And it is giving the output like this:
 ```
Open On-Chip Debugger 0.11.0+dev-02443-gb337b0cfb (2022-12-13-23:37)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Debug: 3 1 command.c:166 script_debug(): command - echo DEPRECATED! use 'adapter speed' not 'adapter_khz'
User : 4 1 command.c:696 handle_echo(): DEPRECATED! use 'adapter speed' not 'adapter_khz'
Debug: 5 1 command.c:166 script_debug(): command - adapter speed 10000
Debug: 6 1 adapter.c:176 adapter_config_khz(): handle adapter khz
Debug: 7 1 adapter.c:140 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 8 1 adapter.c:140 adapter_khz_to_speed(): convert khz to adapter specific speed value
Debug: 9 1 command.c:166 script_debug(): command - echo DEPRECATED! use 'adapter driver' not 'interface'
User : 10 1 command.c:696 handle_echo(): DEPRECATED! use 'adapter driver' not 'interface'
Debug: 11 1 command.c:166 script_debug(): command - adapter driver remote_bitbang
Info : 12 1 transport.c:118 allow_transports(): only one transport option; autoselect 'jtag'
Debug: 13 1 command.c:166 script_debug(): command - echo DEPRECATED! use 'remote_bitbang host' not 'remote_bitbang_host'
User : 14 1 command.c:696 handle_echo(): DEPRECATED! use 'remote_bitbang host' not 'remote_bitbang_host'
Debug: 15 1 command.c:166 script_debug(): command - remote_bitbang host localhost
Debug: 16 1 command.c:166 script_debug(): command - echo DEPRECATED! use 'remote_bitbang port' not 'remote_bitbang_port'
User : 17 1 command.c:696 handle_echo(): DEPRECATED! use 'remote_bitbang port' not 'remote_bitbang_port'
Debug: 18 1 command.c:166 script_debug(): command - remote_bitbang port 9999
Debug: 19 1 command.c:166 script_debug(): command - jtag newtap riscv cpu -irlen 5 -expected-id 0x249511C3
Debug: 20 1 tcl.c:569 jim_newtap_cmd(): Creating New Tap, Chip: riscv, Tap: cpu, Dotted: riscv.cpu, 4 params
Debug: 21 1 tcl.c:593 jim_newtap_cmd(): Processing option: -irlen
Debug: 22 1 tcl.c:593 jim_newtap_cmd(): Processing option: -expected-id
Debug: 23 1 core.c:1472 jtag_tap_init(): Created Tap: riscv.cpu @ abs position 0, irlen 5, capture: 0x1 mask: 0x3
Debug: 24 1 command.c:166 script_debug(): command - jtag names
TAP: riscv.cpu

Debug: 25 1 command.c:166 script_debug(): command - target create riscv.cpu riscv -chain-position riscv.cpu -rtos riscv
Debug: 26 1 target.c:2214 target_free_all_working_areas_restore(): freeing all working areas
User : 27 1 command.c:619 command_run_line(): dm_compliance_test.cfg:18: Error: Unknown RTOS type riscv, try one of: ThreadX, FreeRTOS, eCos, linux, chibios, Chromium-EC, embKernel, mqx, uCOS-III, nuttx, RIOT, Zephyr, hwthread,  or auto
in procedure 'script' 
at file "embedded:startup.tcl", line 26
at file "dm_compliance_test.cfg", line 18
```
should i delete `-rtos riscv` in `target create riscv.cpu riscv -chain-position riscv.cpu -rtos riscv`?
i just want to run a simple test. thanks.

KieenNg, 2024-10-04:
I also encountered the same error. Can you let me know if you have fixed the error or not? @coderalai

phsauter, 2026-08-07:
Closing as an obsolete OpenOCD support question; migration of the example configuration from deprecated -rtos riscv can be tracked separately if desired.