pulp-platform/riscv-dbg #154: `sberror` not clearing properly causes OpenOCD interoperability issues

timothytrippel, 2023-03-29:
The OpenTitan project uses GDB/OpenOCD to develop top-level debug and manufacturing test cases. We have noticed some test failures when performing unaligned system bus accesses (see [here](https://github.com/lowRISC/opentitan/issues/17729#issuecomment-1488923308)).

In short, when an unaligned read operation is performed, `sberror` is set to 3 (which is correct according to the spec), but after, OpenOCD tries to clear the error with a W1C, by sending 0x7, however, the `sberror` field fails to clear. We believe there is an RTL bug described [here](https://github.com/lowRISC/opentitan/issues/17729#issuecomment-1489076212).

Could someone confirm / fix?

bluewww, 2023-03-31:
I agree with @andreaskurth interpretation of the spec. I would opt for the backwards compatible solution because we might already rely on this behavior in some testbench (assuming the meaning of `undefined` is the same as in e.g. the C programming language standard)

andreaskurth, 2023-03-31:
Thanks for your feedback, @bluewww. :+1:

I did not find a definition of _undefined_ in the RISC-V Debug Spec, and other RISC-V specs I know use _implementation-defined_ instead. But I think interpreting _undefined_ as _implementation-defined_ for the Debug Spec, too, makes sense.

I'll create a PR to implement the backwards-compatible solution

closed 2023-04-05 by bluewww

veriquest-verifast, 2026-11-04:
@timothytrippel FYI: this is fixed upstream since v0.8.0 (#155), but PULPissimo (pulp-platform/pulpissimo @ bfc3d9a) still pins riscv-dbg v0.5.1, where the bug is present. VeriQuest (VeriFast Technologies' AI RCA tool) root-caused it there from a failing PULPissimo RTL simulation (Questa) and fixed it with a bitwise write-1-to-clear on sberror; verified in the same simulation.

Fix on v0.5.1: https://github.com/veriquest-verifast/riscv-dbg/tree/fix/sberror-w1c-v0.5.1 (commit 760aac5)

PR #155 (2023-04-05): dm_csrs: Fix W1C behavior of `sberror`
Prior to this PR, the W1C `sberror` field of the `sbcs` CSR only got cleared when `3b'001` was written to it.  When other values, e.g., `3b111`, were written, the field did not get cleared.  This is a mismatch to the RISC-V External Debug Support spec, which states "Writing 1 to every bit clears the field".

This PR fixes that mismatch in a backwards-compatible way, i.e., now any non-zero write to `sberror`, including `3'b001`, clears it.  See issue #154 for further information and discussion.

This resolves #154.

This has been tested in OpenTitan with O [... cut]
commit 60f891b8 (2023-04-05): dm_csrs: Fix W1C behavior of `sberror`

Prior to this commit, the W1C `sberror` field of the `sbcs` CSR only got
cleared when `3b'001` was written to it.  When other values, e.g.,
`3b111`, were written, the field did not get cleared.  This is a
mismatch to the RISC-V External Debug Support spec, which states
"Writing 1 to every bit clears the field".

This commit fixes that mismatch in a backwards-compatible way, i.e.,
now any non-zero write to `sberror`, including `3'b001`, clears it.  See
issue #154 for further information and discussion.

This resolves #154.