pulp-platform/riscv-dbg #180: running_behaviar

maryjohne, 2025-02-07:
Hi 
I am trying to run the project, but i got this:
**Terminal 1:**
        **tb]$ make vcs-run** 
        JTAG remote bitbang server is ready
        Listening on port 9999
        Attempting to accept client socket
**Terminal 2:**
        **tb]$ export JTAG_VPI_PORT=9999**
**Terminal 3:**
        **tb]$ openocd -f dm_compliance_test.cf**
        
        Result:
        ………..
        Info : datacount=2 progbufsize=8
        Info : Examined RISC-V core; found 1 harts
        Info :  hart 0: XLEN=32, misa=0x40001104
        Info : Listening on port 3333 for gdb connections
        Ready for Remote Connections
        Info : Listening on port 6666 for tcl connections
        Info : Listening on port 4444 for telnet connections
**Terminal 4:**
        tb] $riscv32-unknown-elf-gdb prog/test.elf 
        ****From openocd:**
                   Info : accepting 'gdb' connection on tcp/3333
        ****From : gdb** 
                    Reading symbols from prog/test.elf...done.
                    (gdb) 
                    (gdb) target remote localhost:3333
                    Remote debugging using localhost:3333
                    warning: Target-supplied registers are not supported by the current architecture
                    Ignoring packet error, continuing...
                    0x1a000180 in ?? ()
                    (gdb) load 
                    Ignoring packet error, continuing…
                    Loading section .vectors, size 0x100 lma 0x1c000000
                    Ignoring packet error, continuing...
                    ……..
                    Ignoring packet error, continuing...
                    Transfer rate: 1 KB/sec, 5958 bytes/write.
                    (gdb) break main 
                    Ignoring packet error, continuing...
                    Reply contains invalid hex digit 116

**#File: dm_compliance_test.cfg**
              interface remote_bitbang
              remote_bitbang_host localhost
              remote_bitbang_port 9999
              gdb_port 3333
              transport select jtag
              set _CHIPNAME riscv
              jtag newtap $_CHIPNAME cpu -irlen 5 -expected-id 0x249511C3
              set _TARGETNAME $_CHIPNAME.cpu
              target create $_TARGETNAME riscv -chain-position $_TARGETNAME
              riscv set_prefer_sba on
              init
              halt
              echo "Ready for Remote Connections"
              
    **Note: if use riscv64-unknown-elf-gdb prog/test.elf** 
          Reading symbols from prog/test.elf...
          (gdb) target remote localhost:3333
          Remote debugging using localhost:3333
          Ignoring packet error, continuing...
          warning: unrecognized item "timeout" in "qSupported" response
          Ignoring packet error, continuing...
          Remote replied unexpectedly to 'vMustReplyEmpty': timeout
          

**Questions:** 
1) Whats the wrong in my steps? can you help me please?
2) If i run it correctly, what are the tests that i need to use then?

maryjohne, 2025-02-15:
@bluewww can you help me, please?

phsauter, 2026-08-07:
Closing as a general GDB support request without a reproducible defect in this repository.